softbankrobotics-research/qibullet #4: Clarify ROS wrapper usage

Pandhariix, 2019-02-16:
* The requirements of the ROS wrapper should be clarified, and an explanation on how to setup a viable ROS environment should be given.

* If the ROS environments is not properly set up, the __PepperRosWrapper__ class should print the fact that the required ROS libraries are not found, resulting in a failure to call the method __launchWrapper__

Xia0ben, 2019-05-03:
Hi !

Great work you did there, I just launched your simulator, and it started flawlessly !

Although, adhoc "pepper." commands provide the intended results, but I could not succeed in launching the PepperRosWrapper: though the command returns no error, no ROS topics seem to be created. I would gladly assist in writing some docs over the week-end if you could provide a few pointers here =).

This is a bit off-topic, but I saw that you are working on a fork of the naoqi_driver: are there also any plans to update the [pepper_virtual](https://github.com/ros-naoqi/pepper_virtual) package or to link this work with Gazebo in any way ?

Thank you very much for this tool you created, keep up the good work !

Pandhariix, 2019-05-03:
Hi, thanks a lot !

Regarding the __PepperRosWrapper__, I am currently writing a [ROS Tutorial](https://github.com/ProtolabSBRE/qibullet/wiki/Tutorials:-ROS-Interface), i'll try to finish it as soon as possible. 

It makes sense that you weren't able to launch the wrapper: as you noticed we use a custom version of the __naoqi_driver__ repository and of the __naoqi_bridge_msgs__ repositories (that can be found [there](https://github.com/ProtolabSBRE)). You would need to clone the repos in a catkin workspace, compile and source the workspace, and then try to launch the wrapper :) (All of this will be specified in the tutorial)

At the moment, we don't have any plans regarding the pepper virtual package or the Gazebo simulation for Pepper, although I guess that some of the work done in qibullet could be reused to improve those packages (especially the URDF data of the Pepper model)

Thanks for the feedback, glad you like it !

Xia0ben, 2019-05-03:
Waow, that was some reactivity, thanks !
I'll give this a try right now !

Xia0ben, 2019-05-09:
Hey !

I didn't get enough time last friday to finish testing the ROS Wrapper, but got back to it today, and was happy to see you wrote the ROS tutorial, it works great !

I have a few questions but I wonder if it wouldn't be better to open new, specific issues ?

You may want to add in the ROS tutorial the warning that for the moment, only one camera can be subscribed at once ? Do you know what may be the cause for this ?

Wish you a very nice day !

Pandhariix, 2019-05-09:
Hi,

Glad to hear that you managed to get it to work ! You're of course free to add new issues to the repo, but I'm currently thinking about opening a specific chat for the project, on gitter or something.

Yes, and it is also the case if you directly use the Python API of the simulation, only one camera can be subscribed at a time. We based ourselves on the __Bullet Physics Engine__ and on the __Pybullet__ stack to develop __qibullet__, and due to inner workings of those 2 libraries we made that choice, but we might decide to modify that in the future. 

Thanks for pointing that out, I'll add it to the wiki. A nice day to you too !

Xia0ben, 2019-05-09:
Thank you very much for the fast answer !

Okay, I'll wait for a chatroom to be created rather than create new issues for now =). Please ping me when ready !

I made a quick search on why Bullet/PyBullet could only get us one camera at a time but could not find any related material : was your choice motivated by performance reasons ?

Well, I should stop asking questions about this in this issue, since it is not directly related ^^.

See you !

Pandhariix, 2019-05-09:
Hi again,

The link towards the chat should appear at the top of the README. I'll close this issue as soon as I handle the import exception in the ROS wrapper

closed 2019-05-09 by Pandhariix