softbankrobotics-research/qibullet #14: Connect Choregraphe to qibullet

m0rph03nix, 2019-05-09:
Thanks for qibullet. Great job. Is there a way to connect Choregraphe to the naoqi instance of the simulation ? Or in the other way around, is there a way to launch a simulation on an existing naoqi instance. Thank you in advance

m0rph03nix, 2019-05-09:
... I see in the code there is no naoqi behind. :'( So I've got my answer !

mcaniot, 2019-05-09:
Hi ! You can't connect Choregraphe with qibullet. What do you want to do with Choregraphe ? If you want to run some naoqi function on the simulated robot, the qibullet API is closed to the NAOqi one. For example, the function moveTo is implemented like the NAOqi one in qibullet.

m0rph03nix, 2019-05-09:
Yes I realized it in the code just before your answer. Actually my need is not Choregraphe but to run Behaviours with other AL"Stuff" like ALDialog

mcaniot, 2019-05-09:
Alright, I get it. Right now, the simulation is focused on physical interaction and navigation. There is no feature like dialog or human interaction for the moment.

Pandhariix, 2019-05-09:
Like @mcaniot said, we are mostly focused on implementing "low level" features of the NAOqi API (what could could find in ALMotion or in ALVideoDevice for instance).

That being said, you could very well launch a qibullet simulation from a Choregraphe project, and design a wrapper connecting a virtual Choregraphe robot to the qibullet one

m0rph03nix, 2019-05-09:
Thanks ! I will do that and share when I'll find the time.

mbusy, 2020-03-25:
Hi @m0rph03nix, any updates, did you manage to make it work ?

m0rph03nix, 2020-03-27:
Hi @mbusy. I started something last year... but stopped at an early state... lack of time. I'll let you know if I do something more

mbusy, 2020-03-27:
Alright! In that case I'll close the issue for now, feel free to re-open it (or open a new one) if you need specifications or wish to share something new. Take care !

closed 2020-03-27 by mbusy