softbankrobotics-research/qibullet #18: Subscribe/Unsubscribe camera issues

GuillaumeBnd, 2019-06-06:
Our objective was to unsubscribe the pepper camera when the robot is on the move. 
A problem occurs when we first subscribe the pepper camera then take a picture of an object, analyse it in order to return the detection of this object and finally we unsubscribe the camera so that we can move to the next place faster and start again the process with a different object. Unfortunately when the robot is in front of the second object, the picture taken is the one of the first object. We end up with two same picture for two different objects. 

mabye it has something to do with the buffer ?

here is a piece of our code : 

for i in range (3) : 
  pepper.subscribeCamera(PepperVirtual.ID_CAMERA_BOTTOM)       
  img = pepper.getCameraFrame()
  cv2.imshow("bottom camera", img)
  DetctionImage(img)
  pepper.unsubscribeCamera(PepperVirtual.ID_CAMERA_BOTTOM)    
  pepper.moveTo(AxeX, AxeY, 0.0, frame=PepperVirtual.FRAME_ROBOT)

GuillaumeBnd, 2019-06-06:
We kinda solve the problem by take two pictures each times we subscribe the camera and we only analyse the second one so that we are not worried about a false pictures.

Pandhariix, 2019-06-09:
Alright, I see. It might indeed be that the pictured is stored, not updated (because unsubscribeCamera has been called), and thrown out again because there wasn't enough time for the picture to be updated. I'll look into that, in the meantime, could you specify the version of qibullet you're using ?

Pandhariix, 2019-09-02:
The issue is fixed, tested with the following script:
```python
import cv2
import time
import pybullet
import pybullet_data
import numpy as np
from qibullet import SimulationManager
from qibullet import NaoVirtual
from qibullet import PepperVirtual

if __name__ == "__main__":
    images = list()
    simulation_manager = SimulationManager()
    client = simulation_manager.launchSimulation(gui=True)
    pepper = simulation_manager.spawnPepper(client, spawn_ground_plane=True)

    for i in range(3):
        pepper.subscribeCamera(PepperVirtual.ID_CAMERA_BOTTOM)
        images.append(pepper.getCameraFrame())
        pepper.unsubscribeCamera(PepperVirtual.ID_CAMERA_BOTTOM)
        pepper.moveTo(0.1, 0.1, 0.1, frame=PepperVirtual.FRAME_ROBOT)

    simulation_manager.stopSimulation(client)

    prev_img = None

    for i in range(len(images)):
        if prev_img is not None and np.array_equal(images[i], prev_img):
            print("images " + str(i) + " and " + str(i-1) + " are similar")

        cv2.imshow("bottom camera", images[i])
        cv2.waitKey(0)

        prev_img = images[i]

```

closed 2019-09-02 by Pandhariix