softbankrobotics-research/qibullet #20: pybullet.error: cannot import name PoseStampedWithSpeed

undefinedcodezhong, 2019-11-06:
when I run the script "examples/pepper_ros_test.py", I got the error info:
`
Traceback (most recent call last):
  File "pepper_ros_test.py", line 22, in <module>
    wrap = PepperRosWrapper()
  File "/home/zxy/.local/lib/python2.7/site-packages/qibullet/ros_wrapper.py", line 50, in __init__
    raise pybullet.error(MISSING_IMPORT)
pybullet.error: cannot import name PoseStampedWithSpeed
`
So I'd like to know if it's caused by a problem with the naoqi_bridge_msgs version.
Actually I check the source code from "https://github.com/ros-naoqi/naoqi_bridge_msgs" ,I really cannot find the PoseStampedWithSpeed. 
Can you tell me how to fix it ???

mcaniot, 2019-11-06:
Indeed, it's a custom message create by our team. You can fix it by using our fork of the ros packages : [naoqi_bridge_msgs](https://github.com/ProtolabSBRE/naoqi_bridge_msgs) and [naoqi_driver](https://github.com/ProtolabSBRE/naoqi_driver)

undefinedcodezhong, 2019-11-06:
Thank you very much for your prompt reply.
I have downloaded the new packages:naoqi_bridge_msgs and naoqi_driver and compiled them in my ros workspace. Now there's another problem, that is , there exists PoseStampedWithSpeed.msg in the package， but it seem to need the file like xxx.py so that the program can import.
SO now I still cannot import name PoseStampedWithSpeed .
Could you illustrate how to generate the python version package? 
[like in my PC folder : /opt/ros/kinetic/lib/python2.7/dist-packages/naoqi_bridge_msgs， they are .py file , not the .msg file]

undefinedcodezhong, 2019-11-06:
I seem to know. I'll take a  try with genpy to convert the .msg file.

mcaniot, 2019-11-07:
I advise that you remove build and devel in your ros workspace, uninstall ros-kinetic-naoqi-driver and ros-kinetic-naoqi-bridge-msgs (with apt purge if these packages are installed), afterwards you can recompiled your workspace.

Pandhariix, 2019-11-18:
Hi, you should follow the qiBullet wiki, and especially its [ROS Interface section](https://github.com/ProtolabSBRE/qibullet/wiki/Tutorials:-ROS-Interface).

As @mcaniot mentioned, you should start from a clean workspace, remove your build and test folders, clone the latest versions of our forks of [naoqi_driver](https://github.com/ProtolabSBRE/naoqi_driver) and [naoqi_bridge_msgs](https://github.com/ProtolabSBRE/naoqi_bridge_msgs), and compile your workspace.

Remember to source your workspace (_source devel/setup.bash_) before launching _examples/pepper_ros_test.py_, __otherwise it won't work__.

__Warning__: be sure to download the latest version of qiBullet (1.2.1), in order to match the latest version of the naoqi_driver repo.

Let us know how it went!

undefinedcodezhong, 2019-11-25:
yeah， @mcaniot his solution is ok, my problem had been solved!

georgiablanco, 2020-03-30:
Hi I was having the same problem and have done the above and am now getting this error:

[ERROR] [1585572479.266592700]: Client [/pepper_ros] wants topic /move_base_simple/goal to have datatype/md5sum [naoqi_bridge_msgs/PoseStampedWithSpeed/4537c62015e1c2c00d1b2d28bff9f077], but our version has [geometry_msgs/PoseStamped/d3812c3cbc69362b77dc0b19b345f8f5]. Dropping connection.

mbusy, 2020-03-30:
Hi, this issue is a bit old, the qiBullet ROS wrapper now handles the ros-naoqi (official) naoqi-driver, and the softbankrobotics-research fork. A few questions:
* What is your qiBullet version ?
* Did you install naoqi-driver from source or via apt-get ?
* If you installed naoqi-driver from source, which repo did you use ? (ros-naoqi or the softbankrobotics-research fork)

georgiablanco, 2020-03-30:
> Hi, you should follow the qiBullet wiki, and especially its [ROS Interface section](https://github.com/ProtolabSBRE/qibullet/wiki/Tutorials:-ROS-Interface).
>
> As @mcaniot mentioned, you should start [...]

I cloned from the repos linked on this reply and qiBullet version 1.4.0

georgiablanco, 2020-03-30:
I am also launching from this launch file which launches the simulation and rviz

<?xml version="1.0" encoding="UTF-8"?>
<launch>
  <node name="pepper_ros" pkg="turtlebot_explorer" type="pepper_ros_simu.py" output="screen"/>
  <node pkg="rviz" type="rviz" name="rviz_explorer" args="-d $(find turtlebot_explorer)/explorer.rviz"/>
</launch>

georgiablanco, 2020-03-30:
I'm using thi launch file which launches the simulation and rviz

[image]

mbusy, 2020-03-30:
Alright, your version of qiBullet should be able to handle the official and the forked version of the driver.
* The official driver uses basic PoseStamped messages for /move_base_simple/goal
* The forked driver uses PoseStampedWithSpeed messages for /move_base_simple/goal

The PoseStampedWithSpeed message is defined in the forked naoqi_bridge_msgs repository. It seems that your simulation expects PoseStampedWithSpeed message (the qiBullet ROS wrapper detects a forked version of the driver), but receives a PoseStamped one, resulting in that crash.

Do you have publishers on /move_base_simple/goal with a PoseStamped type ? Did you at some point install ROS packages for Pepper (or NAO) on your computer with apt-get ? (From what I understand, your pepper ros wrapper is in pepper_ros_simu.py right ?)

georgiablanco, 2020-03-30:
Yes I had initially installed with apt-get and got the error cannot import name PoseStampedWithSpeed so removed these and followed your instructions above. 

Yes the pepper ros wrapper is in pepper_ros_simu.py

mbusy, 2020-03-30:
Okay I see. The weird thing is that if you installed the ROS packages using apt-get, and used qiBullet 1.4.0, you should have received a warning but not an error (see [here](https://github.com/softbankrobotics-research/qibullet/blob/952829e7f2ea6a7711c168a92f73895fafd70c04/qibullet/ros_wrapper.py#L39)). The warning might be confusing though, I'll clarify it in the next version.
 
> would you advise removing the git clones and reinstalling using apt-get?

It depends on what you want to do. The forked version has a better URDF model of the Pepper robot, so if you want to focus on tasks requiring an accurate geometrical model for Pepper, i would advise you to ensure that you indeed removed all of the packages installed with apt-get, clean and recompile the workspace, and use the forked version. (The RViz config should also be removed)

If you don't care about the accuracy of the geometrical model, and you don't want to recompile everything yourself, removing the clones and using apt-get is definitely the right choice.

georgiablanco, 2020-03-30:
Thank you, I have removed the clone and used apt-get again and it is now working fine just with the warning you have linked :)

mbusy, 2020-03-30:
Great! This warning is a bit confusing, I'll update the message

closed 2020-03-30 by mbusy