softbankrobotics-research/qibullet #26: Pepper meshes/3d model

amoldeshmukh18, 2020-02-18:
Hi,
I am trying to map Qibullet with another software for animation control, I need the exact measurements for Pepper. Is it possible to get a .bvh, or .dae, or .fbx format for 3d model for pepper? Also I tried to unzip meshes.zip but it asks for the password, are we allowed to get one from softbank? Please let me know, thanks
Amol

mbusy, 2020-02-18:
Hi, as specified in the README, the meshes cannot be installed manually, the installer will do that for you. To be able to install the meshes, you will have to agree to the license under which the meshes are published (see [here](https://github.com/softbankrobotics-research/qibullet/blob/master/qibullet/robot_data/LICENSE)). You can find additional information on that topic in the qiBullet [wiki](https://github.com/softbankrobotics-research/qibullet/wiki/Tutorials:-Installation).

Can you give me more details about what you're trying to achieve ?

amoldeshmukh18, 2020-02-27:
Hi,
I wanted to get the exact measurements for the joints as I am mapping the qibullet simulator with another animation controller s/w. I got them my physically measuring them, 
floor -> Hip: 70cm
Hip -> Head (top): 50cm
Hip -> RShoulder: 30cm
floor -> Knee: 35cm
RSHoulder -> RElbow: 20cm
RElbow->RWrist: 15cm
RWrist -> Tip of fingers: 10cm

I still can't move/rotate the wheels, is this not supported by qibullet?

mbusy, 2020-02-27:
If you want the distances between the different links of the Pepper robot (RElbow -> RWrist, etc.) you don't need the meshes, you only need the information provided by the [Pepper URDF](https://github.com/softbankrobotics-research/qibullet/blob/master/qibullet/robot_data/pepper.urdf).

And yes, your can map the movements of a real Pepper robot onto a simulated one, take a look at the [shadowing example](https://github.com/softbankrobotics-research/qibullet/blob/master/examples/pepper_shadowing.py)

amoldeshmukh18, 2020-02-28:
Thanks, the Pepper UDRF file was not easy to decipher to get those measurements, so I used the old fashion measuring tape ;)  I will take a look at shadowing example.

closed 2020-02-28 by amoldeshmukh18