softbankrobotics-research/qibullet #27: Hello,

aadevmobile, 2020-03-16:


aadevmobile, 2020-03-16:
when running the pepper_basic or any other test it loads the model and crashes..
Any ideas?

AA-MacBook-Pro:peppersim aa$ python peper_basic.py 
pybullet build time: Mar 10 2020 14:41:15
Version = 4.1 INTEL-14.2.16
Vendor = Intel Inc.
Renderer = Intel(R) Iris(TM) Plus Graphics 650
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
Fatal Python error: Segmentation fault

Current thread 0x000070000eccd000 (most recent call first):
  File "/Users/aa/Library/Python/2.7/lib/python/site-packages/qibullet/camera.py", line 265 in _getCameraImage
  File "/Users/aa/Library/Python/2.7/lib/python/site-packages/qibullet/camera.py", line 386 in _frameExtractionLoop
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 754 in run
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 801 in __bootstrap_inner
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 774 in __bootstrap

Thread 0x0000000103c45dc0 (most recent call first):
  File "/Users/aa/Library/Python/2.7/lib/python/site-packages/qibullet/camera.py", line 152 in getFrame
  File "/Users/aa/Library/Python/2.7/lib/python/site-packages/qibullet/camera.py", line 304 in _waitForCorrectImageFormat
  File "/Users/aa/Library/Python/2.7/lib/python/site-packages/qibullet/camera.py", line 365 in subscribe
  File "/Users/aa/Library/Python/2.7/lib/python/site-packages/qibullet/robot_virtual.py", line 205 in subscribeCamera
  File "peper_basic.py", line 23 in main
  File "peper_basic.py", line 32 in <module>
Segmentation fault: 11

mbusy, 2020-03-16:
Hi,
Thanks for reporting the issue. Could you specify which version of qiBullet you're currently using, and your OS version ?

aadevmobile, 2020-03-16:
Hi mbusy, 

thank you very much for your fast reply

- macOs Catalina 10.15.1
- qibullet 1.4.0  - numpy 1.16.6  - pybullet 2.6.6

With the sample you provided it doesn't crash but it stops responding within the window; I'm able to press enter in the terminal to exit it. Without the gui = False it seems to run ok.

what are the minimum requirements to run it?  I mean could it be that my mac specs can't handle it?

mbusy, 2020-03-16:
You're welcome. I don't think that your specs are the problem, I believe that the OS doesn't handle OpenGL3 successfully... The way I see it you've got 2 problems:
* The GUI doesn't respond
* A seg fault appears when retrieving camera images

Regarding the first issue, one possible solution would be to replace [that line](https://github.com/softbankrobotics-research/qibullet/blob/952829e7f2ea6a7711c168a92f73895fafd70c04/qibullet/simulation_manager.py#L48) by
```python
pybullet.connect(pybullet.GUI, options="--opengl2")
```

Regarding the second issue, I'm guessing that using pybullet.ER_TINY_RENDERER instead of pybullet.ER_BULLET_HARDWARE_OPENGL ([here](https://github.com/softbankrobotics-research/qibullet/blob/952829e7f2ea6a7711c168a92f73895fafd70c04/qibullet/camera.py#L288)) might fix the issue, but that's not ideal.

I'll test this as soon as possible, but I don't have a system running Mac OS at my disposal (and with the COVID situation I might have some troubles getting my hands on one). I'll eventually test that on a VM

aadevmobile, 2020-03-17:
Hi mbusy,

-  when adding the opengl2 option, GUI doesn't start at all.
-  changing it to ER_TINY_RENDERER I still get the Segmentation fault: 11
same as before :/

mbusy, 2020-03-17:
Alright, it was worth a shot, thanks for testing :smile:.

Another explanation for the first problem could be the simulation step thread. On Windows / Linux, running a simulation with a GUI will automatically spawn an additional thread, stepping the simulation. If I remember correctly, in Mac OS that thread might not be spawned because of OS limitations... So we would need to manually spawn such a thread. We [already do that](https://github.com/softbankrobotics-research/qibullet/blob/952829e7f2ea6a7711c168a92f73895fafd70c04/qibullet/simulation_manager.py#L72) when gui is set to False, in your case doing that when gui is set to True might be relevant

mbusy, 2020-03-17:
This [comment](https://github.com/bulletphysics/bullet3/issues/2014#issuecomment-443380447) (issue in bullet3) seems to confirm it

lubiluk, 2020-11-10:
Hi, I'm also experiencing crashes on MacOS and I'm pretty sure it's related to camera. Without camera use everything seems to work well. For instance this works (https://github.com/softbankrobotics-research/qibullet/blob/master/examples/robot_joint_control.py) while this doesn't (https://github.com/softbankrobotics-research/qibullet/blob/master/examples/pepper_basic.py).

The line that makes everything crash is:
`handle = pepper.subscribeCamera(PepperVirtual.ID_CAMERA_TOP)`

I hope it helps with investigation...

mbusy, 2020-11-10:
Thanks for the feedback @lubiluk, yes the camera is indeed a source of bug on macOS, even when using `ER_TINY_RENDERER ` over `ER_BULLET_HARDWARE_OPENGL` (see previous comment)... 

There is also the problem of the freezing GUI: when you run the `robot_joint_control` example, do you manage to "nudge" (for lack of a better word) the head (or any link) of the robot, by clicking and dragging it with your mouse ?

lubiluk, 2020-11-10:
> There is also the problem of the freezing GUI: when you run the `robot_joint_control` example, do you manage to "nudge" (for lack of a better word) the head (or any link) of the robot, by clicking and dragging it with your mouse ?

Nudging the the robot seems alright. It doesn't freeze on my machine:

```
Version = 4.1 ATI-3.10.18
Vendor = ATI Technologies Inc.
Renderer = AMD Radeon R9 M370X OpenGL Engine
```

elggem, 2020-12-14:
Hi, I looked a bit into this problem and found the following:

There is two crashes happening in the ```_getCameraImage``` function, one for [```pybullet.getLinkState```](https://github.com/softbankrobotics-research/qibullet/blob/46b7f3a6f1672185233bbe5be545a8b176bc9127/qibullet/camera.py#L286) and one for [```pybullet.getCameraImage```](https://github.com/softbankrobotics-research/qibullet/blob/46b7f3a6f1672185233bbe5be545a8b176bc9127/qibullet/camera.py#L308). 

When I replace the first call to ```pybullet.getLinkState``` and assign zero vectors to pos_world and q_world it works until the call to ```pybullet.getCameraImage``` but then segfaults. Changing the renderer to ```ER_TINY_RENDERER``` does not help even in that case the program segfaults.

I dont know how else to proceed to investigate this, but maybe it is helpful to somebody.

mbusy, 2020-12-16:
@elggem Thanks for the info, even though I don't have a lot of time to work on this issue it's always appreciated. The fact that either renderer causes a segfault was already known, but I wasn't aware that `pybullet.getLinkState` also causes a crash, thanks for reporting