softbankrobotics-research/qibullet #31: Gestures

amoldeshmukh18, 2020-04-06:
Hi,
Is it possible to play any gestures from the gesture library for Pepper on qibullet simulator? Please let me know, thanks
Amol

mbusy, 2020-04-07:
Hi,
By gesture library for Pepper, I assume that you mean the [NAOqi animations](http://doc.aldebaran.com/2-5/naoqi/motion/alanimationplayer.html#alanimationplayer) for Pepper ? If so, you cannot directly play the animations on a simulated robot, no. In __qiBullet__, the `Stand`, `StandInit`, `StandZero` and `Crouch` postures can be applied on a virtual Pepper with the `goToPosture` method ([here](https://github.com/softbankrobotics-research/qibullet/blob/952829e7f2ea6a7711c168a92f73895fafd70c04/qibullet/pepper_virtual.py#L332)).

If you want to play an animation on a virtual Pepper, you need to extract the different angular positions of said animation, and apply those positions on the virtual Pepper, with the `setAngles` method.

amoldeshmukh18, 2020-04-07:
@mbusy Thanks for your reply, I wanted to play animations from NAOqi animations, but extracting angular positions and setting these with setAngles method is not very convenient also if want to play animations on the fly using our own generator. Perhaps you can think of adding this functionality in the future updates. Also whats is FPS of the simulator?

mbusy, 2020-04-08:
We probably won't add this functionality, at least in the near future. This simulator aims to provide the "low level" functionalities of the Pepper, NAO, and Romeo robots (such as control and sensing). We don't intend (at least for now) to implement all of the high level features offered by NAOqi. That being said, such a functionality can be coded independently, and used with the simulator.

Regarding the FPS of the simulation:
* When you launch the simulation (with a GUI), a motion thread is spawned. The physics server will run (ideally, if your computer has enough computing power) at 240Hz.
* For the cameras, it also depends on your computer, but you should be limited at 30FPS
* For the GUI render window, between 30 and 60 FPS if I remember correctly

mbusy, 2020-04-24:
@amoldeshmukh18 I will close this issue for now, feel free to reopen it if you have additional questions

closed 2020-04-24 by mbusy