softbankrobotics-research/qibullet #32: ROS Wrapper + NAO

mrsp, 2020-04-12:
Is it possible to read NAO's FSRs and the torso IMU in the simulation?

Thanks in advance

mbusy, 2020-04-12:
Hi,
For now the qiBullet API for NAO doesn't propose such features. But since qiBullet is built on top of pybullet, you could use pybullet methods to retrieve that.

For the torso IMU, you can use the `getLinkState` method:
```python
import pybullet

# qibullet imports and declarations
data = pybullet.getLinkState(
    nao.getRobotModel(),
    nao.link_dict["torso"].getIndex(),
    computeLinkVelocity=1)
```
This will give you the position and the velocity of the torso link, you will have to compute the accelerations yourself though. (Check the return values of the getLinkState method in the [pybullet documentation](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit))

Regarding the FSRs, I would use the `getContactPoints` method (see the [pybullet documentation](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit)), with the `bodyA` parameter being the robot model (`nao.getRobotModel()`) and the `linkIndexA` the index of the link associated to the FSR. You could for instance use the index of the "RFsrRL_frame" link (right foot rear left FSR, retrieve the index with `nao.link_dict["RFsrRL_frame"].getIndex()`).

Although, this link __won't have any contact points__, because of the current URDF. As you can see [here](https://github.com/softbankrobotics-research/qibullet/blob/bcd8c5dea46a6932f1158770c7b58dd31b569985/qibullet/robot_data/nao.urdf#L791), we removed the collision parameters for the NAO's FSRs, to gain more stability. In your case, you need to uncomment all of the collisions for the FSR links, and copy your updated URDF in `~/.qibullet/1.4.0/`

mbusy, 2020-04-12:
Those features are quite relevant, I might eventually implement them in future versions of the simulator, or I'll gladly merge PRs on the subject

mrsp, 2020-04-12:
and quite important for walking!

thanks for the info looking forward to your enhancements

mrsp, 2020-10-23:
Hello any update on the topic? 

Thx

mbusy, 2020-10-27:
Hi @mrsp, no updates on that at the moment

mbusy, 2021-02-05:
@mrsp, IMU and FSRs will be added in the 1.4.3, that should be released pretty soon (check the status of the milestone [here](https://github.com/softbankrobotics-research/qibullet/milestone/3))

closed 2021-02-05 by mbusy

mbusy, 2021-02-08:
@mrsp [release](https://github.com/softbankrobotics-research/qibullet/releases/tag/1.4.3)