softbankrobotics-research/qibullet #33: The ip address of a virtual robot

bornaZbodulja, 2020-04-30:
Hi, 
I'm doing some research regarding navigation solutions for pepper robot. . Because of the current situation I'm not able to work on a real robot so I'm using qibullet as my simulator. I'm trying to use https://github.com/LCAS/spqrel_navigation but for that I need robots ip address. So, my question is does the simulated robot have some kind of an ip address?

Thanks

mbusy, 2020-04-30:
Hi,
The simulated robot don't have any ip address. 

Regarding the SPQREL navigation stack, if they directly use NAOqi to communicate with the robot, I'm afraid that you won't have any easy way to interface the navigation stack with the simulator.

If they use ROS to communicate with the robot (more specifically the packages of the ros-naoqi stack, such as naoqi_driver), you can follow the [qiBullet ROS tutorial](https://github.com/softbankrobotics-research/qibullet/wiki/Tutorials:-ROS-Interface), explaining how to use a qiBullet ROS wrapper. Running a qiBullet virtual robot + a qiBullet ROS wrapper will be equivalent to using a real robot and the ros-naoqi stack, and you won't need any IP address to launch that.

mbusy, 2020-06-02:
Hi @bornaZbodulja, any updates on this issue ?

bornaZbodulja, 2020-06-02:
The navigation via ROS nav stack works fine in simulator, but I wasn't able to use spqrel in simulator because it does not use ROS to communicate with the robot, and I didn't dig into it deeper but is it possible to use gazebo world files in qibullet, because I already have pre built maps of environments in gazebo ?

mbusy, 2020-06-03:
If they don't use ROS, you'll have to replace the qi calls by qiBullet ones (suppressing the need for an IP address).

Regarding the second part of your question, to spawn elements in the simulation you need to use the PyBullet API, and more specifically one (or more) of the following methods:
* `loadURDF`
* `loadSDF`
* `loadMJCF`

If I'm not mistaken, Gazebo worlds are defined in the sdf format, so you could theoretically call the `loadSDF` method on your world file. But I'm not sure that this will yield much results, since the syntax of the world file a bit specific to Gazebo. You might have to individually spawn your components using the aforementioned methods. 
Some additional documentation, the [PyBullet documentation](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.sbnykoneq1me), a [snippet](https://github.com/softbankrobotics-research/qibullet/blob/bcd8c5dea46a6932f1158770c7b58dd31b569985/examples/robot_ros_test.py#L44) calling the `loadURDF` method, and a [snippet](https://github.com/erwincoumans/pybullet_robots/blob/19f4a7d5304dfbaaacbf9a44ddc68b134ef250ab/atlas.py#L11) calling the `loadSDF` method.

We've gone a bit off tracks, I'm going to close this issue for now. Feel free to open a new one if you have additional questions about the simulator

closed 2020-06-03 by mbusy