softbankrobotics-research/qibullet #34: QiSDK Pepper 2.5.* ( API Level 6 ) - compatible?

pzarzycki, 2020-05-14:
Great Project!
I have a kind question regarding compatibility with physical robots (Pepper):
In your documentation it is mentioned "The Python API allowing to interact with a virtual robot is voluntarily close to the one provided by the NAOqi framework  (2-5)"

What about QiSDK API Level 6? Recent versions of Pepper 2.5.*.* (the one we have at the Uni) are no longer accessible via Choreographe. We were using QiSDK for Android so far.
**System version:**
Robot : Robot pepper (AP990397A04P69Mxxxxx)
Type : Release
Version : 2.5.10.7

[image]

mbusy, 2020-05-15:
Thanks!

I'm not sure that I fully understand your question, do you want to know if the qiBullet API will one day be similar to the QiSDK one ? If so, my answer is probably not.

The QiSDK is designed to interact with the robot from an Android activity, and really focuses on human robot interactions / high level behaviours.
qiBullet on the other hand really focuses on "lower level tasks", and can be used to test navigation techniques, grasping techniques, perform reinforcement learning, etc. The simulator is not (at least in its current form) really ideal to perform human robot interaction tasks.

mbusy, 2020-06-02:
Hi @pzarzycki, does that answer your question ?

mbusy, 2020-06-11:
I'll close the issue for now. Feel free to reopen it if you have additional questions on the subject!

closed 2020-06-11 by mbusy