softbankrobotics-research/qibullet #36: Spawning objects in the world

wagenaartje, 2020-06-19:
First of all, thank you for the great work. This is exactly what I needed for a long time.

I would like to spawn objects (e.g. a table and a cube as in your example) into the world, but I could not find how to do this in the documentation. Could you point me to some documentation that explains how to do this?

wagenaartje, 2020-06-19:
I managed to figure it out using one of the examples provided: [Pepper_lasers.py](https://github.com/softbankrobotics-research/qibullet/blob/952829e7f2ea6a7711c168a92f73895fafd70c04/examples/pepper_lasers.py). For example, to spawn a table:

```python
pybullet.setAdditionalSearchPath(pybullet_data.getDataPath())
tableId = pybullet.loadURDF("table/table.urdf",basePosition=[1, 0, 0], globalScaling = 1.0,physicsClientId=client)
```

Makes syre to have imported:

```
import pybullet
import pybullet_data
```

closed 2020-06-19 by wagenaartje

mbusy, 2020-06-19:
You're welcome! You figured it out, currently you have to call the pyBullet API to spawn extra objects in the simulation (you can also use the `loadSDF` and `loadMJCF` methods, see the [pyBullet documentation](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.sbnykoneq1me)). You can find another example in the [robot_ros_test](https://github.com/softbankrobotics-research/qibullet/blob/bcd8c5dea46a6932f1158770c7b58dd31b569985/examples/robot_ros_test.py#L43-L46) example