softbankrobotics-research/qibullet #37: Weird laser behaviour

wagenaartje, 2020-06-20:
I notice that the laser does not always detect objects, it detects objects such as `duck_vhacd` and `sphere8cube` but not `table/table`. Any reason for this behaviour? Is this related to the type of object?

PS: If I run the [pepper_lasers](https://github.com/softbankrobotics-research/qibullet/blob/master/examples/pepper_lasers.py) example and print the value of `laser_list`,  each laser value is always either `0` or `3`. When the lasers go on one by one, they are all `0` and if the lasers are all initialized the values are all `3`. Maybe better to place the duck in a region where it meets the lasers?

wagenaartje, 2020-06-20:
This is the code I am using:

```python
import cv2
import time
import math
import numpy as np
from qibullet import SimulationManager
from qibullet import PepperVirtual

import pybullet
import pybullet_data

def main():
    # Initialize a single simulation
    simulation_manager = SimulationManager()
    client = simulation_manager.launchSimulation(gui=True)

    # Populate the world
    pepper = simulation_manager.spawnPepper(client, spawn_ground_plane=True)
    pybullet.setAdditionalSearchPath(pybullet_data.getDataPath())
    table = pybullet.loadURDF("table/table.urdf",basePosition=[2, 0, -0.05], baseOrientation=[0,0,1,1], globalScaling = 1.0, physicsClientId=client)

    # Initialize the posture
    pepper.goToPosture("Crouch", 0.6)
    time.sleep(1)
    pepper.goToPosture("Stand", 0.6)
    time.sleep(1)
    pepper.goToPosture("StandZero", 0.6)
    time.sleep(1)

    pepper.showLaser(True)
    pepper.subscribeLaser()

    # Start moving
    pepper.moveTo(1, 1, 0, frame=PepperVirtual.FRAME_WORLD, speed=0.1, _async=True)

    try:
        while True:
            # Gather all laser values
            laser_list = pepper.getRightLaserValue()
            laser_list.extend(pepper.getFrontLaserValue())
            laser_list.extend(pepper.getLeftLaserValue())
            print(laser_list)

            cv2.waitKey(1)
    except KeyboardInterrupt:
        simulation_manager.stopSimulation(client)

if __name__ == "__main__":
    main()
```

mbusy, 2020-06-22:
Thanks for the feedback (and for the snippet). You're right, we reduced the laser maximum detection distance to 3 meters, the example has not been updated accordingly, it has to be fixed :)

Regarding the main issue, my guess is that this bug is due to the geometrical properties of the objects... I'll try to run your snippet and check the table in wireframe mode (typing `w` when the simulation is running will enable the wireframe mode)

mbusy, 2020-06-22:
Yep that's what I thought, the table legs are defined from a visual point of view, but not from a collision point of view. The lasers won't be able to detect them
[image]

wagenaartje, 2020-06-23:
Thanks! I didn't know about the wireframe mode

closed 2020-06-23 by wagenaartje