softbankrobotics-research/qibullet #40: How to stop asynchronous movement

wagenaartje, 2020-07-03:
During an asynchronous `moveTo` movement, I want to launch a synchronous movement. This gives the error:

```python
pybullet.error: Already a moveTo asynchronous. Can't launch moveTo synchronous
```

However, is there a way to stop the current asynchronous movement so that a synchronous movement cna be performed?

wagenaartje, 2020-07-03:
Update: I found a work-around for now,

```python
pepper_pos = pepper.getPosition();
pepper.moveTo(0, 0, 0, frame=PepperVirtual.FRAME_ROBOT, speed=1, _async=True)
time.sleep(0.5)
```

Basically, it preforms an asynchronous movement to the current position which is quickly stopped. I'm still curious if there is a cleaner way, for example some kind of `pepper.stop()` function.

mbusy, 2020-07-03:
That's indeed the solution you should use for now. But that's a great point, a `stopMove` methods along the lines of the one defined in the [ALMotion documentation](http://doc.aldebaran.com/2-5/naoqi/motion/control-walk-api.html#ALMotionProxy::stopMove) would be a great add-on

closed 2020-09-22 by mbusy