softbankrobotics-research/qibullet #41: Get position of joint easily

wagenaartje, 2020-07-11:
It would be nice to have a built-in function that could give the position of a join in the world frame. Right now I use the following code to get the joint data:

```python
pybullet.getLinkState(pepper.getRobotModel(), 57)
```

Where `57` corresponds to a joint somewhere in the right hand, which I had to find through a (long) process of trial and error. It would be nice to have a function along the lines of

```python
pepper.getLinkState('RHand')
```

Let me know if there is an easier method than I used (above), if I have some spare time I will try to implement the function.

mbusy, 2020-07-11:
Hi @wagenaartje,

Yep there's an easier way, you don't have to manually find the index of the link. For instance, you could do:
```python
pybullet.getLinkState(
    pepper.getRobotModel(),
    pepper.link_dict["RHand"].getIndex())
```

It could be more straightforward though, having a `getLinkPosition` method in `RobotVirtual`, taking the name of a link as an input and returning its position in the world (translation and orientation) could be a good add-on.

If you do have some spare time and want to implement it, PRs are more than welcome, just make sure that your work is based on develop :wink:

wbadry, 2020-07-23:
Is it possible to get an enumerated list of joint names in NAO? Thanks

mbusy, 2020-07-23:
It is:
```python
# Create the simulation, spawn a virtual nao etc...

# List
joint_names = nao.joint_dict.keys() 

# Enumerate object
enumerated_list = enumerate(nao.joint_dict.keys())
```

closed 2020-09-22 by mbusy