softbankrobotics-research/qibullet #45: Equations for NAO inverse dyanmics

wbadry, 2020-09-19:
Hello,
Is it possible to get the code for NAO inverse dynamics?  I am looking for performing joint control. I would appreciate if there is a function I can get with equations for controlling the locomotion of NAO.

Thank you so much as you always respond in a fast way.

mbusy, 2020-09-21:
Hi, if you're looking for inverse dynamics, my advice would be to directly use the pybullet API, more specifically the `calculateInverseDynamics` method (see the doc of the API [here](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.xfdfclgnw5jo)). The body unique id required by the method corresponds to the model of your robot, you can retrieve it via the `getRobotModel` method of the qibullet API

wbadry, 2020-09-22:
I am lost with the API. Is there any example on NAO such that:

```python
def compute_inverse_dyanmics_nao(x_dot,y_dot):
      .
      .
      .
      .
      return (theta_dot_1, ..... theta_dot_25)

mbusy, 2020-09-22:
What you're trying to achieve is a bit unclear... What is `(x_dot, y_dot)` ? The velocity of the robot's floor projection in the world frame ? and I'm assuming that the theta_dot are the velocities of NAO's joints ?

In any case, if your ultimate goal is to control NAO's walk, we currently don't plan on adding a walk controller to the project (we could consider merging a PR that would add such a feature though). The best course of action for you would be to use an open source walk controller (eg [oru controller](https://github.com/asherikov/oru_walk_module), [mrsp controller](https://github.com/mrsp/nao_walk)), and interface it with a simulated NAO

wbadry, 2020-09-22:
You are absolutely correct. I thought the walking is already implemented but it is possibly a recorded steps if not mistaken . 

Thank you so much @mbusy for your fast response. I'll have a look at what you suggested.

mbusy, 2020-09-22:
No walk controller has been implemented in qibullet. The "NAO walking gif" of the Readme is indeed obtained by replaying/mimicking (in the simulation) the joint positions of a real NAO, walking.

You're welcome, I'll close this issue for now

closed 2020-09-22 by mbusy