softbankrobotics-research/qibullet #46: How to connect via SHARED_MEMORY?

LaiYanKai, 2020-09-30:
Hi,
I've just started Pybullet and Qibullet today, so this may be a silly question: how to connect to the pybullet GUI from another process (nao script?)

I tried this on one py script and managed to run it with the joint parameter example to move the joints for the NAO robot:
```
CLIENT_ID = simulation_manager.launchSimulation(gui=True, use_shared_memory = True)
```

However, when I tried to connect to this GUI from another py script, the following keeps returning -1, which means it can't connect:
```
pybullet.connect(pybullet.SHARED_MEMORY)
```

I read from [image] that shared_memory is a feature? Does this mean that it is not supported?

I'm running on Python 2.7, Ubuntu 18.04.5...

mbusy, 2020-09-30:
Hi,

I assume that you're refering to that part of the [wiki](https://github.com/softbankrobotics-research/qibullet/wiki/Tutorials:-Setup#launching-and-stopping-a-simulation)?

The `use_shared_memory` parameter will only be taken into account when `gui=False` (see that [block of code](https://github.com/softbankrobotics-research/qibullet/blob/46b7f3a6f1672185233bbe5be545a8b176bc9127/qibullet/simulation_manager.py#L47-L74)). The goal of the `use_shared_memory` parameter is to allow you to parallelize the motion server of headless simulation instances (without a GUI), and therefore running them faster (check the plot and the code samples related to that feature in the wiki).

What you're trying to do seems different, from what I understand you're looking to connect a new client to an already existing pybullet server (with a GUI). Am I correct ? That is currently not proposed in qiBullet, but it's definitely something doable, either by using the pybullet API directly, or by modifying the [`launchSimulation`](https://github.com/softbankrobotics-research/qibullet/blob/46b7f3a6f1672185233bbe5be545a8b176bc9127/qibullet/simulation_manager.py#L27) method of the `SimulationManager` class in qiBullet. This could even be introduced as a new feature

LaiYanKai, 2020-10-01:
Hi,
I didn't know the wiki existed, thanks for sharing.
Also, thanks for sharing the code part. Just as I thought, it could be headless at the moment to speed up testing. I will modify the code as suggested, it's easier haha. I would close this issue at this moment, because it shouldn't be too difficult to do it. I would try it out some time next week.

Thanks for the informative answer! :D

closed 2020-10-01 by LaiYanKai