softbankrobotics-research/qibullet #47: Simple walking module for NAO robot

ritwik-j, 2020-10-23:
Hi All,

I am planning to use the qibullet environment to render a NAO robot in a maze-like environment and perform Reinforcement Learning using OpenAI for path planning and navigation.There is a gif of walking NAO robot in the main page of this repo: 

[image]

Could the creators make available the code of this walking robot, or the source for the same?

Thank you for the help!

mbusy, 2020-10-23:
Hi @ritwik-j, as mentioned in #45:
> No walk controller has been implemented in qibullet. The "NAO walking gif" of the Readme is indeed obtained by replaying/mimicking (in the simulation) the joint positions of a real NAO, walking.

and 
> In any case, if your ultimate goal is to control NAO's walk, we currently don't plan on adding a walk controller to the project (we could consider merging a PR that would add such a feature though). The best course of action for you would be to use an open source walk controller (eg [oru controller](https://github.com/asherikov/oru_walk_module), [mrsp controller](https://github.com/mrsp/nao_walk)), and interface it with a simulated NAO

ritwik-j, 2020-10-24:
Dear @mbusy, thank you for the quick response. I will check out the two walking modules that you have shared.

ritwik-j, 2020-10-26:
Dear @mbusy, just a quick follow up question. Is it possible to interface the Qibullet engine with Choregraphe? I understand that Choregraphe has a move function which could possibly be used for this purpose.

mbusy, 2020-10-27:
The connection to Choregraphe was discussed in issue #14: for your specific issue, the problem would be that the move function of Choregraphe is actually going to call the move function of NAOqi (that will in turn, for NAO, use the walk controller).

The qiBullet API mimics the NAOqi API, but qiBullet does not depend on NAOqi, there is no direct way to connect the two. There might be a way to indirectly reach your goal though (detailed in issue #14):
>That being said, you could very well launch a qibullet simulation from a Choregraphe project, and design a wrapper connecting a virtual Choregraphe robot to the qibullet one

The mentioned wrapper would get the joint positions of the NAO in Choregraphe at a specific frequency, and set corresponding joint positions on the qiBullet NAO (very similar to the [shadowing example](https://github.com/softbankrobotics-research/qibullet/blob/master/examples/pepper_shadowing.py) with Pepper). But that's not an optimal solution at all.

ritwik-j, 2020-10-29:
Understood. I think what I am looking for is more suitable for ROS rather than PyBullet engine perhaps. Thank you for your help!

mbusy, 2020-10-29:
You're welcome ! I'm closing the issue for now

closed 2020-10-29 by mbusy

robogeekcanada, 2021-10-14:
Hi thank you for developing `qibullet`. I really enjoy how well thought out and organized the code is. Just small contribution for Nao robot walking. I got the parameters from Webots site and was able to do a bunch of motions. Here is the link if someone is interested: https://github.com/robogeekcanada/pybullet-tutorials/blob/main/nao_motions.py

mbusy, 2021-10-18:
Thanks a lot @robogeekcanada, that's very nice to hear! Thanks for sharing that, I'll eventually take a look if I have the time. In m
the meantime, if you're interested to contribute directly to qibullet by adding the walk we can open an issue to discuss about the format of that contribution ;)