softbankrobotics-research/qibullet #48: meshes_installer_27.pyc not installable

Mara01010011, 2020-10-28:
I am using Linux Mint 20 with Kernel: Linux 5.4.0-52-generic and Python 2.7.18

When executing any test program with an IDE (pycharm), python2.7 says 
`The robot meshes and URDFs will be installed in the /home/mara/.qibullet/1.4.0 folder. You will need to agree to the meshes license in order to be able to install them. Continue the installation (y/n)? `
As soon as I insert y, it's sending an Error message:
```
Installing the meshes and URDFs in the /home/mara/.qibullet/1.4.0 folder...
Python 2.7 detected
E
======================================================================
ERROR: setUpClass (__main__.PepperBaseTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/mara/Masterarbeit/Software/qibullet/tests/base_test.py", line 27, in setUpClass
    spawn_ground_plane=True)
  File "/home/mara/.local/lib/python2.7/site-packages/qibullet/simulation_manager.py", line 145, in spawnPepper
    pepper_virtual = PepperVirtual()
  File "/home/mara/.local/lib/python2.7/site-packages/qibullet/pepper_virtual.py", line 36, in __init__
    tools._install_resources()
  File "/home/mara/.local/lib/python2.7/site-packages/qibullet/tools.py", line 145, in _install_resources
    import meshes_installer_27 as meshes_installer
ImportError: Bad magic number in /home/mara/.local/lib/python2.7/site-packages/qibullet/robot_data/installers/meshes_installer_27.pyc

----------------------------------------------------------------------
Ran 0 tests in 13.655s

FAILED (errors=1)
```
When executing any test program in my terminal, the gui starts, but I can't see any robot mesh.
When I execute this: `$ sudo python meshes_installer_27.pyc `
There is this error message. `RuntimeError: Bad magic number in .pyc file`

mbusy, 2020-10-29:
Hi, quick question, did you install the project with pip or from sources ? (seems like a git LFS problem, see [this part](https://github.com/softbankrobotics-research/qibullet/wiki/Tutorials:-Installation#from-source) of the wiki)

Mara01010011, 2020-10-29:
I used 
`pip install --user qibullet`
But I installed pip in a different way:
```
curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py
python get-pip.py
```
What do you mean by "git LFS problem"

mbusy, 2020-10-29:
By "git LFS problem", I'm referring to what is explained in the wiki section (installation from source) that I linked:
>You will encounter a bad magic number error when installing the extra resources if git-lfs isn't correctly installed. To solve that error, install git-lfs, go the the repository folder, and type in git lfs pull. This command will download the lfs resources, you can then resume installing qiBullet.

But since you installed qiBullet via pip, you shouldn't get that issue. I already relaunched a [CI job](https://travis-ci.org/github/softbankrobotics-research/qibullet/jobs/729290834) (Ubuntu Xenial, Python 2.7.15), it manages to install the project (and the meshes) from source, and to run the unit tests.

I will try to reproduce your bug, keep me posted if you have more information

mbusy, 2020-11-02:
@Mara01010011, I can't reproduce your problem... 

You should encounter the bad magic number issue if the .pyc file is not a valid pyc, or if your Python version doesn't match the Python version used to generate the file (only the major and minor version number should count).
* I checked the pyc files that have been exported to PyPi (for the last qiBullet version), they are valid.
* `meshes_installer_27.pyc` was created with Python 2.7.12 and you're using Python 2.7.18, the version major and minor are the same, it should be compatible. I still wanted to make sure of that, so I launched a CI with different versions of Python 2.7: everything seems to work for Python 2.7.18 (see [here](https://travis-ci.org/github/softbankrobotics-research/qibullet/jobs/740901305))

I don't really understand why you run into that issue. The pyc file might be corrupted, did you try to uninstall everything, remove the extra resources (in `/home/mara/.qibullet/`), and reinstall qiBullet? Or to install from source ?

Mara01010011, 2020-11-08:
I uninstalled qiBullet and pyBullet and reinstalled it with pip. There is an error message, when I run a python program, that should spawn nao:
```
pybullet build time: Oct  8 2020 00:08:00
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=Intel
GL_RENDERER=Mesa Intel(R) HD Graphics 620 (KBL GT2)
GL_VERSION=4.6 (Core Profile) Mesa 20.0.8
GL_SHADING_LANGUAGE_VERSION=4.60
pthread_getconcurrency()=0
Version = 4.6 (Core Profile) Mesa 20.0.8
Vendor = Intel
Renderer = Mesa Intel(R) HD Graphics 620 (KBL GT2)
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = Intel
Workaround for some crash in the Intel OpenGL driver on Linux/Ubuntu

The qibullet ressources are not yet installed.

Installing resources for qiBullet

The robot meshes and URDFs will be installed in the /home/mara/.qibullet/1.4.0 folder. You will need to agree to the meshes license in order to be able to install them. Continue the installation (y/n)? ven = Intel
Workaround for some crash in the Intel OpenGL driver on Linux/Ubuntu
XIO:  fatal IO error 62 (Timer expired) on X server ":0.0"
      after 1964 requests (1964 known processed) with 0 events remaining.
```

mbusy, 2020-11-08:
Okay, that's different from your original issue, you didn't run (at least not yet) into the bad magic number problem.

The `Workaround for some crash in the Intel OpenGL driver on Linux/Ubuntu` issue is actually mentioned in the [troubleshooting](https://github.com/softbankrobotics-research/qibullet#troubleshooting) section of the README. Is your GPU an Intel HD Graphics 620, or do you have something else ?

mbusy, 2020-12-28:
@Mara01010011 any updates on the issue ?

Mara01010011, 2021-01-04:
> @Mara01010011 any updates on the issue ?

Sorry, no. I tried to reinstall it a few times in different ways, but it didn't work. But I made every adjustment of my python program for the nao with our real nao in the laboratory, so at the moment I am not in need of qibullet. 
I think, the problem lies within my linux installation. I reinstalled it but sustained the /home patition. Since then I sometimes experience permission issues mostly with applications which use java. So the problem in this case might be a user permission problem, too.

mbusy, 2021-01-04:
Alright. I'll close the issue for know, you can re-open it if you run into this problem again.

closed 2021-01-04 by mbusy

giodegas, 2021-07-03:
with `git-lfs pull` I solved the same issue with Python3.8 , thank you @mbusy