softbankrobotics-research/qibullet #49: Joint_trajectory action not implemented

AmelieLegeleux, 2020-11-12:
Hi,
I'm trying to use the code I have written with real robot with QiBullet. With QiBullet, I cannot use /joint_trajectory action like I did with real robot (with the same ros package /naoqi_driver). I can move the robot with /joint_angles but not with /joint_trajectory. Why?
Thank you for your help!

mcaniot, 2020-11-13:
Hi,
It's normal because it's not linked in the qibullet ros_wrapper. The ros_wrapper contained the basic topics linked as the /joint_angles topic (see `_jointAnglesCallback` in [ros_wrapper.py](https://github.com/softbankrobotics-research/qibullet/blob/master/qibullet/ros_wrapper.py)). If you want, you can implement the function to link  /joint_trajectory and request a PR.

AmelieLegeleux, 2020-11-13:
Ok, thank you for you answer. I have used /joint_angles but now I prefer to use the action joint_trajectory to define the entire movement. When there are a lot of positions for a single movement, the movement is more smooth with an action (avoiding publishing multiple positions) espacially with real robot.