softbankrobotics-research/qibullet #50: Manually stepping the simulation

lubiluk, 2020-11-16:
Hi, please consider allowing for manual simulation stepping. It would be useful for faster-than-real-time simulations.
Currently, this can only be done in a hacky way with GUI and shared_memory modes:

```
pybullet.setRealTimeSimulation(0, physicsClientId=self._client)
...
pybullet.stepSimulation(physicsClientId=self._client)
```
Unfortunately DIRECT mode (which is the fastest one) does not allow for that trick as it always spawns a spinning thread for simulation stepping.

mbusy, 2020-11-16:
I agree, the user should be able to step manually in Direct mode (very useful to obtain deterministic behaviours). Some calls will have to be avoided while running qiBullet with the "DIRECT manual stepping" combo though, for instance a call to a synchronous `moveTo`.

mbusy, 2020-11-16:
This [commit](https://github.com/softbankrobotics-research/qibullet/commit/dc8a9c0bbd15be6ff0c4a64847d3a06d72b29f8e) should allow you to manually step in DIRECT mode. It will be added to the qibullet package in the [next release](https://github.com/softbankrobotics-research/qibullet/milestone/3).
```python
manager = SimulationManager()
client = manager.launchSimulation(gui=False, auto_step=False)
# stuff
manager.stepSimulation(client)
# more stuff
manager.stopSimulation(client)
```

lubiluk, 2020-11-16:
Great news. 
I would like to make a case for supporting manual stepping in GUI mode as well. In Reinforcement Learning (for which I use qibullet) it is desirable to visually examine learned policies by running them in the same way as during the training in DIRECT mode. That is, by manually stepping the simulation.

mbusy, 2020-11-17:
Having that manual stepping in GUI mode could also be interesting (again especially for the purpose of being deterministic), I don't know if I'll add it to the simulation before the release of the 1.4.3 though (if you want to submit a PR adding it to qiBullet I'll review it). If manual stepping is allowed in GUI mode, it should also be allowed when using shared memory ([here](https://github.com/softbankrobotics-research/qibullet/blob/46b7f3a6f1672185233bbe5be545a8b176bc9127/qibullet/simulation_manager.py#L63)).

To circle back to your original comment, bear in mind that using "faster-than-real-time" simulation instances might give you weird physics behaviours (depending on the content of your environment, your simulation parameters etc). If your goal is to transfer your learnt policies on a real robot, that would render them pretty much useless. If you stay in simulation though (with the same parameters) it will work

lubiluk, 2020-11-18:
#51

closed 2020-11-18 by mbusy