softbankrobotics-research/qibullet #56: How to get absolute position of link/part of Robot?

czp16, 2020-11-25:
I want to leverage z-axis coordinate (height) of head or other part to know whether the robot falls down or not during simulating, but function `getlinkstate` (it seems a pybullet function in `RobotVirtual.getLinkPosition`) seems to only return relative position. (When NaoRobot goes to "Stand" posture, the z-axis coordinate of "Head" returned by `getlinkstate` < 0.2, so I think it must be a relative position)

Is there any way to get absolute position or ensure whether the robot falls?

Thanks!

mbusy, 2020-12-02:
Hi, as you pointed out, `getLinkPosition` uses the `getlinkstate` function from pybullet, and focuses on the first two elements returned by the method, respectively the translation and the orientation of the center of mass of the link in the world frame (directly using the position of the URDF link frame makes more sense, I'll probably update that). What is your qiBullet version ?

When I tried the method on the head of a simulated NAO, I obtain a value of 0.56 on z, which makes sense given the height of the real robot (58cm)... Here is the code I used (robot joint control example with a few modifications):
```python
#!/usr/bin/env python
# coding: utf-8

import sys
import time
import pybullet as p
from qibullet import SimulationManager
from qibullet import PepperVirtual
from qibullet import NaoVirtual
from qibullet import RomeoVirtual

if __name__ == "__main__":
    simulation_manager = SimulationManager()

    if (sys.version_info > (3, 0)):
        rob = input("Which robot should be spawned? (pepper/nao/romeo): ")
    else:
        rob = raw_input("Which robot should be spawned? (pepper/nao/romeo): ")

    client = simulation_manager.launchSimulation(gui=True)

    if rob.lower() == "nao":
        robot = simulation_manager.spawnNao(client, spawn_ground_plane=True)
    elif rob.lower() == "pepper":
        robot = simulation_manager.spawnPepper(client, spawn_ground_plane=True)
    elif rob.lower() == "romeo":
        robot = simulation_manager.spawnRomeo(client, spawn_ground_plane=True)
    else:
        print("You have to specify a robot, pepper, nao or romeo.")
        simulation_manager.stopSimulation(client)
        sys.exit(1)

    time.sleep(1.0)
    joint_parameters = list()

    for name, joint in robot.joint_dict.items():
        if "Finger" not in name and "Thumb" not in name:
            joint_parameters.append((
                p.addUserDebugParameter(
                    name,
                    joint.getLowerLimit(),
                    joint.getUpperLimit(),
                    robot.getAnglesPosition(name)),
                name))

    while True:
        for joint_parameter in joint_parameters:
            robot.setAngles(
                joint_parameter[1],
                p.readUserDebugParameter(joint_parameter[0]), 1.0)
            print("translation: " + str(robot.getLinkPosition("Head")[0]))
            time.sleep(1)
```
Do you still have a z value inferior to 0.2 when the robot is standing ?

czp16, 2020-12-03:
Thanks for your response! I think I made some mistakes when customizing simulation env and now it works.

closed 2020-12-03 by czp16