softbankrobotics-research/qibullet #60: Support for Python 3.9

lubiluk, 2020-12-28:
I've already checked that the simulator works just fine on Python 3.9. 
Unfortunately, mesh installer doesn't support it yet. Looks like it would be a minor change but it involves some bits that I believe are distributed in an opaque .pyc version.

mbusy, 2020-12-28:
Yup, that is indeed a good idea. For license purposes, the installer has to be delivered as a pycs (1 for each Python version), I'll generate it for Python 3.9 when I have the time

lubiluk, 2021-01-08:
Sorry, I realised I may have done one change to make this work under Python 3.9. 
`robot_module.py:58`
`isAlive -> is_alive`

mbusy, 2021-02-01:
Fixed with the following [commit](https://github.com/softbankrobotics-research/qibullet/commit/858ba781cf44a34e366f9c850a9c0b6f303cebc7)

closed 2021-02-01 by mbusy