softbankrobotics-research/qibullet #67: Installation Process robot meshes and URDFs

SutirthaChakraborty, 2021-04-04:
Hi, I am new to this. I used pip to install the library. From where shall I download the robot meshes and URDFs? I see the robots didn't spawn on the environment.
This might be a basic question. Sorry for that. 
Thanks,
Suti

mbusy, 2021-04-13:
Hi Suti, I'm not so sure that I understand your question:
> How can I use a video to feed into the Nao's Camera

By default, the camera of the virtual NAO is a simulated one, and will return images captured in the simulation (see [here](https://github.com/softbankrobotics-research/qibullet/wiki/Tutorials:-Virtual-Robot#sensing-with-a-virtual-robot) for more information). From what I understand you want to stream your own video data in the simulation ?

> and its music in the microphone sensor?

The answer to this one is easier :smile: The simulated robot doesn't have microphones or speakers, so that's not possible at the moment

SutirthaChakraborty, 2021-04-13:
Yes, I want to stream my video data. How to create the environment for that?
Will there be any sooner release with audio data as well? Or any other alternative to achieve that?

mbusy, 2021-04-14:
I don't think that there is a nice and easy way to do that, you could always try to create a wall (that would act as a screen), segment your video into successive images and change the texture of the wall to match each image, one after the other. (that's quite dirty though)

Regarding the audio, we probably won't add that to the simulator, at least not soon, since the audio is not handled by the pybullet APIs.

What are you actually trying to achieve by streaming that into the simulation? Depending on your goal, using another simulator (such as morse or webots) might be more relevant

SutirthaChakraborty, 2021-04-14:
Thank you for your response. I am trying to simulate an orchestra environment [Video Link](https://www.youtube.com/watch?v=bPwA4gVobJ0). I want the humanoid bot to react with this.

mbusy, 2021-04-14:
Ok, and I guess that you want the robot to capture that to take into account the specificities of its sensors (eg camera fov, resolution, mic sensitivity etc) before processing the video and the audio?

SutirthaChakraborty, 2021-04-14:
Yes

mbusy, 2021-04-14:
Alright, I see! The camera of the robot will have a correct fov, resolution and intrinsic parameters, but there will still be a reality gap between the simulated camera and the real one (the real camera will render noisier images).

My guess is that if you really want to take into account the model of the sensors, you should maybe use a different simulator, that handles that kind of stuff more accurately (you can take a look at [ignition](https://ignitionrobotics.org/features), or [webots](https://cyberbotics.com/doc/guide/sensors#camera-sensors)). I think that there is already a NAO robot model available in webots

SutirthaChakraborty, 2021-04-14:
Okay, Thank you so much for all your suggestions. I will have a look at them.

closed 2021-04-14 by SutirthaChakraborty