softbankrobotics-research/qibullet #76: pepper_shadowing

ysys98, 2022-03-17:
when I rosrun pepper_shadowing.py
In fact, I replaced pepper with Nao, and then this result occurred. 
At first, I thought it could not be replaced, 
but this result also occurred when I ran the code before replacement
[image]
what should I do ？

mbusy, 2022-03-17:
Hi @ysys98, you can indeed replace Pepper with NAO in the example, more specifically by modifying [this line](https://github.com/softbankrobotics-research/qibullet/blob/fd9284072221a87801ba1ab4490d2fdbb15bc933/examples/pepper_shadowing.py#L37). But the example doesn't use ROS, it directly uses the qi SDK to interact with the robot (as you can see [here](https://github.com/softbankrobotics-research/qibullet/blob/fd9284072221a87801ba1ab4490d2fdbb15bc933/examples/pepper_shadowing.py#L27-L32)). I would advise to launch the program without rosrun:
```bash
# replace the x.x.x.x by the ip of the real robot
python pepper_shadowing.py --ip x.x.x.x
```

You probably got the error message because you didn't specify the ip of your robot

ysys98, 2022-03-17:
> Hi @ysys98, you can indeed replace Pepper with NAO in the example, more specifically by modifying [this line](https://github.com/softbankrobotics-research/qibullet/blob/fd9284072221a87801ba1ab4490d2fdbb15bc933/examples/pepper_shadowing.py#L37). But the example doesn't use ROS, it directly uses the qi SDK to interact with the robot (as you can see [here](https://github.com/softbankrobotics-research/qibullet/blob/fd9284072221a87801ba1ab4490d2fdbb15bc933/examples/pepper_shadowing.py#L27-L32)). I would advise to launch the program without rosrun:
> 
> ```shell
> # replace the x.x.x.x by the ip of the real robot
> python pepper_shadowing.py --ip x.x.x.x
> ```
> 
> You probably got the error message because you didn't specify the ip of your robot
@mbusy thank you a lot ！
and I am a novice student, so thank you for taking the trouble to solve these very simple problems for me. I just saw you in Naoqi_ Driver's reply to me, which is what bothers me. I downloaded qibullet according to the tutorial, so I have to say that Ubuntu already exists ROS_ Wrapper.py，And I checked it
[image]
, all I do is run the code like examples, but I run the robot_ ros_ test. After py, whether I run rosrun rviz rviz directly on the terminal 
[image]
or I run the launch file that starts rviz in vscode，I can't find odmo in the Fixed Frame. I'm quite confused
[image]

ysys98, 2022-03-18:
> Hi @ysys98, you can indeed replace Pepper with NAO in the example, more specifically by modifying [this [...]

@mbusy The same problem also appears in Pepper , when I run the robot_ ros_ test. py, load rviz through the launch file. Similarly, there is no Odom in the Fixed Frame in pepper's rviz, and there is a base_ footprint in Fixed Frame , I remember that base_footprint link should be  written in pepper.urdf
[image]

ysys98, 2022-03-19:
@mbusy Thank you very much! I have followed your reply to do things about  [https://github.com/ros-naoqi/naoqi_driver/issues/147](url)
when I  added _ broadcastodometry method into robot_ ros_ test. py, I made an error after the operation. I checked and found that I may not have a publisher about odometry, so I found a odometry_ publisher.cpp program that might provide a publisher  at [http://docs.ros.org/en/kinetic/api/odometry_publisher_tutorial/html/odometry__publisher_8cpp_source.html](url)
, I think I have one parameter of _broadcastodometry (def _broadcastodometry (self, odometry_publisher):), but when I run odometry_ publisher. cpp to start the publisher of odometry and run the robot_ ros_ test. py I modified, but he automatically closes it and outputs such a log:
Traceback (most recent call last):
  File "/home/roskinetic/demo01_ws/src/qibullet/scripts/robot_ros_test.py", line 50, in <module>
    wrap._broadcastOdometry("odom_pub")
  File "/home/roskinetic/.local/lib/python2.7/site-packages/qibullet/ros_wrapper.py", line 208, in _broadcastOdometry
    physicsClientId=self.robot.getPhysicsClientId())
TypeError: function takes at most 2 arguments (3 given)
Stopping the ROS wrapper: function takes at most 2 arguments (3 given)
[robot_state_publisher_roskinetic_VirtualBox_21246_1218137726150742084-1] killing on exit
[WARN] [1647703861.872514]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
Stopping roslauncher
This is the odometry_ publisher.cpp I found, the robot_ ros_ test. py I modified, and his input log：
[image]
[image]

[image]
[image]

[image]
[image]

ysys98, 2022-03-19:
> Hi @ysys98, you can indeed replace Pepper with NAO in the example, more specifically by modifying [this [...]
@mbusy Thank you very much! I have followed your reply to do things about  [https://github.com/ros-naoqi/naoqi_driver/issues/147](url)
when I  added _ broadcastodometry method into robot_ ros_ test. py, I made an error after the operation. I checked and found that I may not have a publisher about odometry, so I found a odometry_ publisher.cpp program that might provide a publisher  at [http://docs.ros.org/en/kinetic/api/odometry_publisher_tutorial/html/odometry__publisher_8cpp_source.html](url)
, I think I have one parameter of _broadcastodometry (def _broadcastodometry (self, odometry_publisher):), but when I run odometry_ publisher. cpp to start the publisher of odometry and run the robot_ ros_ test. py I modified, but he automatically closes it and outputs such a log:
Traceback (most recent call last):
  File "/home/roskinetic/demo01_ws/src/qibullet/scripts/robot_ros_test.py", line 50, in <module>
    wrap._broadcastOdometry("odom_pub")
  File "/home/roskinetic/.local/lib/python2.7/site-packages/qibullet/ros_wrapper.py", line 208, in _broadcastOdometry
    physicsClientId=self.robot.getPhysicsClientId())
TypeError: function takes at most 2 arguments (3 given)
Stopping the ROS wrapper: function takes at most 2 arguments (3 given)
[robot_state_publisher_roskinetic_VirtualBox_21246_1218137726150742084-1] killing on exit
[WARN] [1647703861.872514]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
Stopping roslauncher
This is the odometry_ publisher.cpp I found, the robot_ ros_ test. py I modified, and his input log：
[image]
[image]

[image]
[image]

[image]
[image]

ysys98, 2022-03-21:
> Hi @ysys98, you can indeed replace Pepper with NAO in the example, more specifically by modifying [this [...]

@mbusy When I checked, I found that vscode did not display the import pybullet module. I thought it might be that it was not installed properly during installation, so I reinstalled it again and solved this problem.
 For the above problems, I'm made a small change for ros_ wrapper. py , 
[image]
Oh, by the way, I changed the parameters above wrap._broadcastOdometry("odom_pub") from odometry_publisher to odom_pub,
[image]
After I ran it, although it did not close automatically, an error log still appeared
[image]

closed 2022-03-23 by mbusy